sekdiy/FlowMeter #1: reset() => getTotalVolume()

w0nk1, 2018-01-02:
Wouldn't it make sense to clear getTotalVolume() either during the reset() method?

sekdiy, 2018-01-03:
Question: in what scenario would this make sense for you and what's your intended use case?

The persistence of the total volume counter across resets is intentional.

The basic idea is that of a meter – in contrast to a simple counter.
Comparable to the speedometer in a car, the flow meter has two counters:

- current value(s) inbetween resets,
- total value(s) across resets.

Resetting the total value(s) wouldn't make much sense, since they would then never differ from the current value(s) in most scenarios.

Ideally, the total volume and flow rate would persist across power cycles (in order to make tampering more difficult). 
But I don't consider this a responsibility of the FlowMeter class at this time.

I could add a way to store these values into non-volatile memory if there is demand for that.
Pull request are of course welcome.

w0nk1, 2018-01-11:
In my special case I'm using it for a beverage machine with a bunch of meters. In this case I use a shift register to switch between those meters and need a variable to keep track of this. So it would be nice, to have a value which counts the poured liquid and is resettable.

sekdiy, 2018-01-12:
Can you clarify how you intend to use the FlowMeter object in this?
Normally, you would just create multiple FlowMeter instances.
How you store them (e.g. in an indexed array) is up to your application.

I don't think you should count the flow rates of multiple sensors using just one FlowMeter object.
But if you must do it this way, why not use the `getCurrentVolume()` method instead?
The 'current' values will by zeroed by calling `reset()`.

w0nk1, 2018-01-20:
Thank you! :-) I'll check on that!

sekdiy, 2020-08-12:
I'd like to reopen this issue in order to find a solution to the problem of wether to reset _total_ values on a call to `reset()` or not.

Since the FlowMeter library would lose a feature – total values strictly being monotonic (i.e. never losing volume that has been accounted for) during runtime – I'm thinking about adding a third set of values to `get` instead.

The meaning would be:
- get volume and average flow rate since last reset
- accumulate the same way total flow rate and volume are accumulated
- reset to zero on every call of `reset()`

I think it would be rather simple to accomplish this, but I can't come up with a good name right now. 
It's not a set of `current` values, but also not a set of `total` values.
It's inbetween, but the only name I could come up with so far is `getVolumeSinceLastReset()`...

Issues affected: #13, #21

tasosman, 2021-12-13:
Hey how can I reset the total volume?I like to the amount of ml it push and after with one button I want to reset it to 0 and I start the measurements again .

sekdiy, 2021-12-15:
Hey @tasosman,

could you achieve your goal by using current volume instead of total volume?

Current volume – and other 'current' values – are affected by a `reset()`, because they are intermittent (only relevant for one flow event).

Total volume – and other 'total' values – are persistable values, i.e. for bookkeeping or long-term summation/statistics.

tasosman, 2021-12-15:
Yes I check it when I use current volume is sawing l/s but me use the total so I can see how much ml run ,but I fix that I use what are you take off of your code so I put it back and now I can reset the total volume by using the reset .it is a little only difficult to calibrate my sensor now because I use coffee machine and is passing max 400 ml/minute

sekdiy, 2021-12-15:
Alright, thanks for the feedback.

Yes, modifying the code to also `reset` the `total` values is what most people do when they have an application like that.

Well, this thread/issue serves as a place for me to discuss this as a feature request.

So I consider your case another data point in favour of the feature. :)

tasosman, 2021-12-15:
> Coffee machines (I'm a LaCimbali Junior user) were one of my initial motivations to create this library, along with garden irrigation.
>
> If you'd like to discuss your sensor and its calibration, please open another issue here.

Come on bro ,xaxaxaxa same passion here of course I have made 3 custom espresso machine and this is one of my last project

Best regards
Anastasios Manolidis

tasosman, 2021-12-15:
I will open new one

sekdiy, 2022-12-09:
I'm closing this issue due to inactivity.
If you have new information, please reopen it.

closed 2022-12-09 by sekdiy