sekdiy/FlowMeter #2: Compile for ESP8266?

helipilot04, 2018-01-10:
Very new to programming.    Thanks for this great library, I used it to make a great flow rate meter with totalizer.   I wanted to re build the sketch and compile it for use with a Huzzah esp8266 module and send data to my Adafruit Io panel..   I have successfully complete this task with other sensors, but the flowmeter.cpp file is throwing me for a loop.  The .cpp file seems to drop many issues for the Esp8266 compiling.  Adruino IDE 1.8.1 
Any help would be appreciated.  hopefully the code is attached.  

```
/* hookup instructions::  
Compiled for Arduino NANO V3.1  rebuilt for Adafruit Huzzah ESP8266
The meter were using is a FS300A
meter pins:  Black to GND, Red to +5v, yellow or signal to D2
reset switch pins: one side of momentary switch to GND the other to D4
power in :  gnd and + 5v 
Displays: Adafruit .56" 7 segment displays with backpack.  I2C 
Display I2C pins:  VCC - +5v, GND to gnd, SCL to A5 and SDA to A4  -------*/

// Libraries
#include <ESP8266WiFi.h>
#include "Adafruit_MQTT.h"
#include "Adafruit_MQTT_Client.h"
#include <Wire.h>
#include <Adafruit_GFX.h>
#include "Adafruit_LEDBackpack.h"
Adafruit_7segment matrix1 = Adafruit_7segment();
Adafruit_7segment matrix2 = Adafruit_7segment();
#include <FlowMeter.h>                 

// WiFi parameters
#define WLAN_SSID       "***enter here*****"
#define WLAN_PASS       "***enter here**********"

// Adafruit IO
#define AIO_SERVER      "io.adafruit.com"
#define AIO_SERVERPORT  1883
#define AIO_USERNAME    "**username**"
#define AIO_KEY         "**user key**"

// define the sensor characteristics here
const double cap = 60.0f;  // l/min
const double kf = 5.5f;     // Hz per l/min
const int pin = 2;         // INT0 pin on most Arduinos
const int ResetButton = 4;      //pin that reset switch is attached to
int ResetNow;

// let's provide our own sensor properties (without applying further calibration)(FS300A)
FlowSensorProperties MySensor = {60.0f, 5.5f, {1, 1, 1, 1, 1, 1, 1, 1, 1, 1}};

// now initialize the flow meter with the pin and sensor settings
FlowMeter Meter = FlowMeter(pin, MySensor);

// set the measurement update period to 1s (1000 ms)
const unsigned long period = 1000;
float FlowrateGallons = 0;
float VolumeGallons = 0;

// define an 'interrupt service handler' (ISR) for every interrupt pin you use
void MeterISR() {
  // let our flow meter count the pulses
  Meter.count();
} 

// Functions
void connect();
void publish();
long time = 0;
long sendTime = 30000;

// Create an ESP8266 WiFiClient class to connect to the MQTT server.
WiFiClient client;

// Setup the MQTT client class by passing in the WiFi client and MQTT server and login details.
Adafruit_MQTT_Client mqtt(&client, AIO_SERVER, AIO_SERVERPORT, AIO_USERNAME, AIO_KEY);

// Setup feeds for publishing data  totalgallons and flowrate
Adafruit_MQTT_Publish totalGallons = Adafruit_MQTT_Publish(&mqtt, AIO_USERNAME "/feeds/basement.total-gallons");
Adafruit_MQTT_Publish flowRate = Adafruit_MQTT_Publish(&mqtt,  AIO_USERNAME "/feeds/basement.flow-rate");

/*************************** Sketch Code ************************************/

void setup() {
  // Connect to WiFi access point.
  Serial.println(); Serial.println();
  delay(10);
  Serial.print(F("Connecting to "));
  Serial.println(WLAN_SSID);

  WiFi.begin(WLAN_SSID, WLAN_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(F("."));
  }
  Serial.println();

  Serial.println(F("WiFi connected"));
  Serial.println(F("IP address: "));
  Serial.println(WiFi.localIP());

  // connect to adafruit io
  connect();
  
  pinMode(ResetButton, INPUT_PULLUP);
  ResetNow = HIGH;
  
  // prepare serial communication
  Serial.begin(115200);
  matrix1.begin(0x70);
  matrix2.begin(0x72);    //  (0x74) if need be
  
  // enable a call to the 'interrupt service handler' (ISR) on every rising edge at the interrupt pin
  // do this setup step for every ISR you have defined, depending on how many interrupts you use
  attachInterrupt(pin, MeterISR, RISING);

  // sometimes initializing the gear generates some pulses that we should ignore
  Meter.reset();
  
}

void loop() {
  // ping adafruit io a few times to make sure we remain connected
  if(! mqtt.ping(3)) {
    // reconnect to adafruit io
    if(! mqtt.connected())
      connect();
  }
  
  // wait between output updates
  delay(500);
 
  // process the (possibly) counted ticks
  Meter.tick(period);

   // convert system variables to Gallons  .264 gallon / litre
  FlowrateGallons = (Meter.getCurrentFlowrate() / 3.78541178);
  VolumeGallons = (Meter.getTotalVolume() / 3.78541178);

  // output result to serial monitor for debugging
  // uncomment next 2 lines to see results in serial monitor
  //Serial.println("Currently " + String(Meter.getCurrentFlowrate()) + " l/min, " + String(Meter.getTotalVolume())+ " l total.");
  //Serial.println("Currently " + String(FlowrateGallons) + " G/min, " + String(VolumeGallons)+ " Gal total.");

  // if a preset time has elapsed then publish the data to adafruit io
  if(millis() - time > sendTime) {
    publish();
    time = millis();
  }

  // print result to 7 segment display
  matrix1.setBrightness(10);
  matrix1.print(FlowrateGallons);
  matrix1.writeDisplay();
  delay(10);
  matrix2.setBrightness(5);
  matrix2.print(VolumeGallons);
  matrix2.writeDisplay();
  
  // reset button function, will reset meter if pressed
  ResetNow = digitalRead(ResetButton);
   if(ResetNow == LOW){
    Meter.reset();
   }
}

// Publish data to adafruit IO
void publish() {
  if (! totalGallons.publish(VolumeGallons))
    Serial.println(F("Failed to publish Total Gallons"));
  else
    Serial.println(F("Total Gallons published!"));
    Serial.println(VolumeGallons);
    
  if (! flowRate.publish(FlowrateGallons))
    Serial.println(F("Failed to publish Flow Rate"));
  else
    Serial.println(F("Flow Rate published!"));
    Serial.println(FlowrateGallons);
}

// connect to adafruit io via MQTT
void connect() {

  Serial.print(F("Connecting to Adafruit IO... "));

  int8_t ret;

  while ((ret = mqtt.connect()) != 0) {

    switch (ret) {
      case 1: Serial.println(F("Wrong protocol")); break;
      case 2: Serial.println(F("ID rejected")); break;
      case 3: Serial.println(F("Server unavail")); break;
      case 4: Serial.println(F("Bad user/pass")); break;
      case 5: Serial.println(F("Not authed")); break;
      case 6: Serial.println(F("Failed to subscribe")); break;
      default: Serial.println(F("Connection failed")); break;
    }

    if(ret >= 0)
      mqtt.disconnect();

    Serial.println(F("Retrying connection..."));
    delay(5000);

  }

  Serial.println(F("Adafruit IO Connected!"));

}
```

sekdiy, 2018-01-10:
Thanks for reporting this issue.

As far as I'm aware, this library has not been ported to the ESP architecture (or existing ports haven't been made public yet).

There are a lot of things the `#include`d library `ESP8266WiFi.h` is handling differently from the original `Arduino.h` library.

I can spot two different things within your compiler output:

- different min()/max() functions between Arduino and ESP8266 (https://github.com/esp8266/Arduino/issues/398#issuecomment-356034907)
- undeclared SREG register name in ESP8266

Unfortunately I won't find the time to port this library at the moment (although I have ESPs available).

So this is your chance to fix the errors yourself and improve the FlowMeter library. ;)

sekdiy, 2018-01-10:
# Different min()/max() functions between Arduino and ESP8266

Could you please replace the expression `min(decile, 9)` (in line 49) by `min(decile, (unsigned int) 9)` and report back?

helipilot04, 2018-01-10:
I will give it a try...   Thanks for the feedback.  At least I now have a direction to go!!   I had already fixed the SREG, but took out the fix so you could see all the issues..

sekdiy, 2018-01-10:
## Dndeclared SREG register name in ESP8266

Alright, the SREG register isn't explicitly used on the ESP, but you could probably do without it (as it is just a compatibility measure on Arduino).

Could you please completely remove the lines containing `SREG` (if not already done) and add a followup `sei()` for every `cli()` (i.e. lines 45 and 71 in the unmodified FlowMeter.cpp)?

sekdiy, 2018-01-10:
See esp8266/Arduino/issues/615

helipilot04, 2018-01-10:
Below is the New improved FlowMeter.cpp
Compiled fine for the Adafruit Huzzah ESP8266, just need to run the sketch and see if the output makes sense..  
Thanks again, changes were easy as pie for you I see..  Prob would have taken me a while to fix this.

FlowMeter.cpp
```
/*
 * Flow Meter
 */

// Compatibility with the Arduino 1.0 library standard
#if defined(ARDUINO) && ARDUINO >= 100
#include "Arduino.h"
#else
#include "WProgram.h"
#endif

#include "FlowMeter.h"                                                      // https://github.com/sekdiy/FlowMeter
#include <math.h>

FlowMeter::FlowMeter(unsigned int pin, FlowSensorProperties prop) :
    _pin(pin),                                                              //!< store pin number
    _properties(prop)                                                       //!< store sensor properties
{
  pinMode(pin, INPUT_PULLUP);                                               //!< initialize interrupt pin as input with pullup
}

double FlowMeter::getCurrentFlowrate() {
    return this->_currentFlowrate;                                          //!< in l/min
}

double FlowMeter::getCurrentVolume() {
    return this->_currentVolume;                                            //!< in l
}

double FlowMeter::getTotalFlowrate() {
    return this->_totalVolume / (this->_totalDuration / 1000.0f) * 60.0f;   //!< in l/min
}

double FlowMeter::getTotalVolume() {
    return this->_totalVolume;                                              //!< in l
}

void FlowMeter::tick(unsigned long duration) {
    /* sampling and normalisation */
    double seconds = duration / 1000.0f;                                    //!< normalised duration (in s, i.e. per 1000ms)
    
    cli();                                                                  //!< going to change interrupt variable(s)
    sei();	 
    double frequency = this->_currentPulses / seconds;                      //!< normalised frequency (in 1/s)
    this->_currentPulses = 0;                                               //!< reset pulse counter after successfull sampling
    
    /* determine current correction factor (from sensor properties) */
    unsigned int decile = floor(10.0f * frequency / (this->_properties.capacity * this->_properties.kFactor));  //!< decile of current flow relative to sensor capacity
    this->_currentCorrection = this->_properties.kFactor / this->_properties.mFactor[min(decile, (unsigned int)9)];           //!< combine constant k-factor and m-factor for decile

    /* update current calculations: */
    this->_currentFlowrate = frequency / this->_currentCorrection;          //!< get flow rate (in l/min) from normalised frequency and combined correction factor
    this->_currentVolume = this->_currentFlowrate / (60.0f/seconds);        //!< get volume (in l) from normalised flow rate and normalised time

    /* update statistics: */
    this->_currentDuration = duration;                                      //!< store current tick duration (convenience, in ms)
    this->_currentFrequency = frequency;                                    //!< store current pulses per second (convenience, in 1/s)
    this->_totalDuration += duration;                                       //!< accumulate total duration (in ms)
    this->_totalVolume += this->_currentVolume;                             //!< accumulate total volume (in l)
    this->_totalCorrection += this->_currentCorrection * duration;          //!< accumulate corrections over time
}

void FlowMeter::count() {
    this->_currentPulses++;                                                 //!< this should be called from an interrupt service routine
}

void FlowMeter::reset() {
    
    cli();                                                                  //!< going to change interrupt variable(s)
    sei();
    this->_currentPulses = 0;                                               //!< reset pulse counter
    

    this->_currentFrequency = 0.0f;
    this->_currentDuration = 0.0f;
    this->_currentFlowrate = 0.0f;
    this->_currentVolume = 0.0f;
    this->_currentCorrection = 0.0f;
    this->_totalVolume = 0.0f;
}

unsigned int FlowMeter::getPin() {
    return this->_pin;
}

unsigned long FlowMeter::getCurrentDuration() {
    return this->_currentDuration;                                          //!< in ms
}

double FlowMeter::getCurrentFrequency() {
    return this->_currentFrequency;                                         //!< in 1/s
}

double FlowMeter::getCurrentError() {
    /// error (in %) = error * 100
    /// error = correction rate - 1
    /// correction rate = k-factor / correction
    return (this->_properties.kFactor / this->_currentCorrection - 1) * 100;  //!< in %
}

unsigned long FlowMeter::getTotalDuration() {
    return this->_totalDuration;                                            //!< in ms
}

double FlowMeter::getTotalError() {
    /// average error (in %) = average error * 100
    /// average error = average correction rate - 1
    /// average correction rate = k-factor / corrections over time * total time
    return (this->_properties.kFactor / this->_totalCorrection * this->_totalDuration - 1) * 100;
}

FlowSensorProperties UncalibratedSensor = {60.0f, 5.0f, {1, 1, 1, 1, 1, 1, 1, 1, 1, 1}};
FlowSensorProperties FS300A = {60.0f, 5.5f, {1, 1, 1, 1, 1, 1, 1, 1, 1, 1}};
FlowSensorProperties FS400A = {60.0f, 4.8f, {1, 1, 1, 1, 1, 1, 1, 1, 1, 1}};

```

helipilot04, 2018-01-10:
FYI,  after testing, all communications and setup worked as intended..  Thanks for the great library and help.

sekdiy, 2018-01-10:
Great to hear it's working four you now.

Please note that the call to `sei()` should follow after accessing the variable(s), not directly after the call to `cli()`.

My wording was a bit unclear.

This might not be a problem in your project, as likely no other interrupts affect the variable access anyway. But you never know... ;)

I'll update the source file accordingly.

sekdiy, 2018-01-10:
Done. Please report any further issues or incompatibilities, thanks!

closed 2018-01-10 by sekdiy

rhyslbw, 2018-02-26:
@sekdiy Thanks for porting this to ESP ( It's also working on ESP32)

Could you bump the version so Platform.io picks up the change?

sekdiy, 2018-02-27:
Done. :)