sekdiy/FlowMeter #6: Connecting multiple flow meters

highergroundstudio, 2018-10-14:
I tried to connect two flow meters on pins 2 & 3 of an Uno but it only shows the total for the first meter when polled using `String(Meter1.getTotalVolume())` 

I used the simple example and set it up like this: 
`FlowMeter Meter1 = FlowMeter(2);
FlowMeter Meter2 = FlowMeter(3);`

sekdiy, 2018-10-16:
Hi, do you experience this behaviour only with `getTotalVolume()` or do other methods (like `getTotalFlowrate()`, `getCurrentVolume()` and `getTotalVolume()`) also behave that way?

highergroundstudio, 2018-10-16:
Just tried it and all methods behave that way

sekdiy, 2018-10-16:
The simulator link seems to respond with a "missing sketch" error.

Thanks for the minimal example.

Let's take a look at lines 10 through 15:
```cpp
// define an 'interrupt service handler' (ISR) for every interrupt pin you use
void MeterISR() {
  // let our flow meter count the pulses
  Meter1.count();
  Meter2.count();
}
```

The comment in line 10 states that your project requires one interrupt service routine (ISR) handler per interrupt pin you use.
For two interrupt pins that makes two ISR handlers.

In lines 13 and 14 you count the pulses for both Meter obects _at the same time_. 
It seems logical that they would then end up having the same flow and volume statistics over time.

Could you please try separating the two calls to `count()` into two ISR handlers like so:
```cpp
void Meter1ISR() {
  Meter1.count();
}

void Meter2ISR() {
  Meter2.count();
}
```
and then attaching them to the interrupt pins like so:
```cpp
  attachInterrupt(INT0, Meter1ISR, RISING);
  attachInterrupt(INT1, Meter2ISR, RISING);
```
and report your findings? Thanks.

highergroundstudio, 2018-10-16:
That works perfectly then! Thank you so much!! It may be a good idea to add this as an example ;)

highergroundstudio, 2018-10-16:
`#include <FlowMeter.h>                  // https://github.com/sekdiy/FlowMeter

// connect a flow meter to an interrupt pin (see notes on your Arduino model for pin numbers)
FlowMeter Meter1 = FlowMeter(2);
FlowMeter Meter2 = FlowMeter(3);

// set the measurement update period to 1s (1000 ms)
const unsigned long period = 1000;

// define an 'interrupt service handler' (ISR) for every interrupt pin you use
void Meter1ISR() {
  // let our flow meter count the pulses
  Meter1.count();
}

// define an 'interrupt service handler' (ISR) for every interrupt pin you use
void Meter2ISR() {
  // let our flow meter count the pulses
  Meter2.count();
}

void setup() {
  // prepare serial communication
  Serial.begin(9600);

  // enable a call to the 'interrupt service handler' (ISR) on every rising edge at the interrupt pin
  // do this setup step for every ISR you have defined, depending on how many interrupts you use
  attachInterrupt(INT0, Meter1ISR, RISING);
  attachInterrupt(INT1, Meter2ISR, RISING);

  // sometimes initializing the gear generates some pulses that we should ignore
  Meter1.reset();
  Meter2.reset();
}

void loop() {
  // wait between output updates
  delay(period);

  // process the (possibly) counted ticks
  Meter1.tick(period);
  Meter2.tick(period);

  // output some measurement result
  Serial.println("M1 " + String(Meter1.getCurrentFlowrate()) + " l/min, " + String(Meter1.getTotalVolume())+ " l total.");
  Serial.println("M2 " + String(Meter2.getCurrentFlowrate()) + " l/min, " + String(Meter2.getTotalVolume())+ " l total.");
}
`

closed 2018-10-16 by highergroundstudio

sekdiy, 2018-10-17:
Great. And done. :)

sekdiy, 2023-01-26:
Hi @Yabi23,

is this in relation to the original question in this issue by highergroundstudio?

If so, did you try to reproduce highergroundstudio's issue?

If not, please open a new issue, this issue is resolved and closed since many years.

Thank! :)