sekdiy/FlowMeter #7: Split up flow sensor and flow meter into different classes and files.

sekdiy, 2018-10-17:
Currently the sensing and the metering functions are merged into one class/object, requiring users to deal with the complexity of the metering interface even if all they need is a simple flow rate value. 

When connecting one or multiple sensors, users should have the choice between a lightweight interface to their hardware and a sophisticated metering solution.

kapyaar, 2018-10-29:
Hi, Another nice feature would be to add support for sensors with Voltage (analog) and Current (4-20mA) signals in addition to the pulse type.

sekdiy, 2018-10-30:
Hi @kapyaar, can you recommend any analog frontends and sensors for that purpose? 
Traditionally, current loops are rather difficult to glue to general purpose Arduino/ESP/uC boards. 

I don't have access to voltage or current loop flow sensors, so personally I'm not planning for 10V or 4-20mA compatibility at this time. 

But it sure would be nice to have. :)

sekdiy, 2020-08-13:
Closed due to inactivity.

closed 2020-08-13 by sekdiy