sekdiy/FlowMeter #11: How I can change sensor properties in execution Time

hardmax, 2019-04-30:
Hi,

I review that exist a class FlowSensorCalibration() to change properties, but How can I use it??
How Can I change properties in the setup or loop??

How can I do it if I got 2 sensors?

sekdiy, 2019-05-14:
Hi @hardmax,

I understand that you would like to change/adjust sensor properties at run time, i.e. after the initial setup and/or inbetween measurements. 

The `FlowSensorCalibration` class is a convenience/helper class, it mainly provides assistance during the *manual* calibration process. 
Its main purpose so far is to contain all the scalar values and provide a `FlowSensorProperties` struct for the `FlowMeter` class.
But I introduced this class with future improvements in mind. 

Could you please describe the scenario in which you'd like to change the properties of a sensor at run time? 
So far this library assumes a constant sensor that doesn't change its properties inbetween (manual) calibration runs. 

As of yet, there's no support for automated calibration.
So maybe your issue is a feature request to introduce this capability.

It wouldn't be too complicated to allow the library to update the sensor properties at runtime, but I'd like to understand your intended use case before changing anything.

hardmax, 2019-05-17:
Hi,

Thanks for your answer and your work..they are amazing.

Well, we use in ours projects a professional flowmeter and they are calibrated in fabric. All flowmeter has its own k factor (same model but they have his own k factor) and this k factor is different with different fluids.

We need change k factor because in our application, users use different sensor and fluids...

I think put k factor in no volatile memory, but gif to users opportunity to change it when they change fluid or sensor.

For now I declare empty variables and in a function I put values(read from epprom), but I don't know if when i call that function in a second time this going to work.....

Thanks again for you work.

Regards

sekdiy, 2019-05-21:
Thanks for the kind feedback, always appreciated. :)

The way I understand your use case, you don't actually want (your customers) to go through the calibration process every time.

Instead, you assume that your calibration *profiles* (i.e. k-factors, m-factors, etc) are good.
You just want (your customers) to be able to apply them dynamically. 

Maybe what you need isn't an improved calibration process, but a kind of profile management?

Also, if you don't mind, may I ask what type of flow sensors/meters you use?

sekdiy, 2019-07-18:
If you have new information, please reopen this issue.

closed 2019-07-18 by sekdiy