sekdiy/FlowMeter #12: Software Debounce

mrarmyant, 2019-07-16:
I’m using EKM flow meters for a project that use a reed switch for triggering the pulses from the wheel. It would be nice to be able to supply a debounce interval to work with reed switches over hall sensors.

sekdiy, 2019-07-18:
Thanks for that, would certainly make for a cool feature.

Okay, I understand that your meters use mechanical reed switches as opposed to (filtered) semiconductor hall sensors. 

It might sound lazy, but do you have a way to debounce in hardware, i.e. via a simple RC filter? :)

Other than that, your best option at this point probably will be to debounce within the ISR handler routine (see e.g. https://forum.arduino.cc/index.php?topic=45000.msg325940#msg325940 for a working example).

I don't see a robust way to provide this as part of the FlowMeter library, since everyone's metering application and use case is different.
One person's bounce is another person's signal... 

So as part of the library this would call for a tuning option with fool proof defaults and presets, which I consider impossible for everyone. 

Again, not to sound lazy, but nothing beats debouncing in hardware...

sekdiy, 2019-07-18:
Here's a nice tutorial I just found (after a brief search) about debouncing hardware interrupt signals in both hardware and software: https://www.jeremyblum.com/2011/03/07/arduino-tutorial-10-interrupts-and-hardware-debouncing

closed 2020-01-08 by sekdiy