sekdiy/FlowMeter #13: Meter.reset() function not reset totalvolume counter

AvinashPudale, 2019-09-30:
**Description**
A clear and concise description of what the bug is.

**Reproduction**
Steps to reproduce the behavior:
1. Connect '...'
2. Measure '...'
3. Error '...'

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware (please complete the following information):**
 - Board type
 - Sensor type
 - Pins used

**Software (please complete the following information):**
 - OS
 - IDE
 - Other libraries

**Minimal example to reproduce (please complete the following information):**
 - Code to reproduce
 - Wiring to reproduce

**Additional context**
Add any other context about the problem here.

sekdiy, 2019-10-04:
This behaviour is by design, see #1 .

The reset functionality should not void total volume count, because a *meter* should be tamper proof.

If you want to reinitialize the meter, please do so by explicitly setting the values (see #1).

If you think that this should be improved, please give some context and insight.

AvinashPudale, 2019-10-09:
void FlowMeter::reset() {
    cli();                                                                  //!< going to change interrupt variable(s)
    this->_currentPulses = 0;                                               //!< reset pulse counter
    sei();                                                                  //!< done changing interrupt variable(s)
   **this->_totalVolume = 0.0f;**                                            //just added this line now works as expected 
   this->_currentFrequency = 0.0f;
  i added one line at above fuction now works as expected by me

sekdiy, 2019-10-09:
Yes, as I mentioned above, this possibility was discussed in #1. 

But do you think this is a feature that's missing in the library?
If so, why?

AvinashPudale, 2019-10-10:
When we call reset() function in setup function it work like charm. 
 I use reset() function in loop with if statement ,to  trigger relay on or off based on total volume despensed from trigger. 
 Then I  use reset() function. Which  reset all current volume not totalvalume. 
Total volume count remains as privous value.

sekdiy, 2019-10-10:
Actually, that's what `current volume` is intended for.

The difference between `current volume` and `total volume` is that the former is resettable, the latter is not.
This is by design. 

If you intend to trigger a relay based on volume since last reset, isn't that `current volume`?
In other words: isn't your modification unifying `current` and `total` volumes, just making them identical all the time?

sekdiy, 2020-01-08:
Closed due to inactivity.

closed 2020-01-08 by sekdiy