sekdiy/FlowMeter #14: M Factor String, How to assing to which interval, what Mfactor needs to be used.

Ameengray, 2019-11-07:
**Description**
I am using an 10mm flow sensor connected to a tap line, the water flow is controlled by the Tap, However the tap is supper sensitive and reaches its max in a 40% turn.  The max flowrate i am able to record is 5.5L /pm.  We are able to measure the the max flowrate by tweaking our K-factor as a start, hence the nominal is just right, then i go about measuring different flowrates for the mfactor values. We have now 7 intervals ( 7 M-factors values, we start with 1 for all 10 Mfactor values and do actual/nominal to the right mfactor values )  

How do we decide which interval value should  correspond to the M factor value ?
When we put the first 6 values as we have and 1 for remaining, its not giving us right output. 
We tried different combinations, its no results. What would be the logic behind how we place the Mfactor values ? 

I am able to get the updated values, but dont know how to how place them correctly in the Mfactor String. 

Could anyone perhaps shed some light on this ?

sekdiy, 2019-11-22:
Let's try the [Step-by-step procedure](https://github.com/sekdiy/FlowMeter/wiki/Calibration#step-by-step-procedure) from the Calibration page in order to get on the same page.

1. Software: I'm assuming you're using the [_Calibration example_](https://github.com/sekdiy/FlowMeter/blob/master/examples/Calibration/Calibration.ino). 
1. Setup: What `capacity` and `kFactor` did you use as initial values, just so that I can comprehend. 
1. Bucket: What size are you using?

Maybe you can disclose the sensor you'r using, so that I can follow your calculations.

sekdiy, 2020-01-08:
Closed due to inactivity.

closed 2020-01-08 by sekdiy

uribeer, 2021-01-11:
I want to make sure I understand the M factors setting correctly, can you tell me if the following statements are correct?

1. You use only one parameter for upper boundary (capacity) so the lower boundary is always 0
2. If my sensor has a capacity of 5 l/min but I use it only up to 3 l/min I can set the capacity to 3 and be more accurate with the M factors
3. If I use capacity of 3 then the M factors cover these ranges:
 0-0.3, 0.3-0.6, 0.6-0.9 .... So I actually need to measure the flowrate at 0.15, 0.45, 0.75 to be the most accurate

sekdiy, 2021-01-16:
Someone asked me:

> I want to make sure I understand the M factors setting correctly, can you tell me if the following statements are correct?

So I thought I'll just answer here quickly:

> 1. You use only one parameter for upper boundary (capacity) so the lower boundary is always 0

Yes. Capacity really is a scalar property. The usable range is usually smaller, the linear range even more so. 

> 2. If my sensor has a capacity of 5 l/min but I use it only up to 3 l/min I can set the capacity to 3 and be more accurate with the M factors

Yes. Should your flow rate surpass the 3 l/min, the library will just assume the upper boundary

https://github.com/sekdiy/FlowMeter/blob/6c8be63e6656b2fdc06868393b69d51fd3b36251/src/FlowMeter.cpp#L54  

> 3. If I use capacity of 3 then the M factors cover these ranges: 0-0.3, 0.3-0.6, 0.6-0.9 .... So I actually need to measure the flowrate at 0.15, 0.45, 0.75 to be the most accurate

Yes. I would assume that the linear approximation used should work that way, although I haven't double-checked if that's what the code really does. :D