sekdiy/FlowMeter #18: Verify sensor calibration value

thijstriemstra, 2020-05-22:
I wanted to [calibrate the sensor](https://github.com/sekdiy/FlowMeter/wiki/Calibration) but it's waaay to technical for me unfortunately.

My flow meter comes with this graph: http://www.microelectronicos.com/datasheets/UCTS0058.pdf

[image]

Could you tell me what a reasonable calibration value is based on this graph? Or does it not work like that?

thijstriemstra, 2020-05-22:
I did a test with the default settings using 5 liters and it was off by almost 4 liters:

```
Currently 14.00 l/min, 8.94 l total.
```

Now on to calibrating..

thijstriemstra, 2020-05-22:
I ended up using this for now:

```cpp
const double cap = 30.0f;  // l/min
const double kf = 11.4f;   // Hz per l/min
FlowSensorProperties sensorConfig = {cap, kf, {1, 1, 1, 1, 1, 1, 1, 1, 1, 1}};
_sensor = new FlowMeter(_pin, sensorConfig);
```

but love to be able to calibrate it more (e.g. get rid of those 1's).

sekdiy, 2020-05-23:
Kudos for attempting to calibrate! :) 

The diagram ends at a flow rate of `9 l/min`, but the true capacity is determined by how much you're able put through it without overwhelming it. 

Did you really push `30 l/min` through it? 
That would be _a lot_ for such a flow sensor, the impeller can start to tumble above it's _true_ capacity, which would in turn lead to missed pulses. 
 
You can actually calculate the `k-factor` from the diagram. 
At `2 l/min` it shows slightly above `20 Hz`, thus about `11 Hz per l/min`. 
At `9 l/min`it shows roughly `110 Hz`, thus roughly `12 Hz per l/min`. 
So, if you ended up with a `k-Factor` of `11.4 Hz per l/min`, you're pretty spot on with respect to the diagram. 

Your sensor's datasheet has a table of 'electrical characteristics' that says **[4.1Q] ±10%** (in line 8, 'Flow pulse characteristics'). 
I would assume that this figure is intended to mean what I call `k-Factor`, which would indicate that the datasheet is inconsistent!

sekdiy, 2020-05-23:
> but love to be able to calibrate it more (e.g. get rid of those 1's).

They're actually helping you with that (if we want to trust the diagram again for a moment). 

The diagram has a table showing corresponding flow rates `l/min` and pulse frequencies in `Hz`.

You can just divide them in order to get a starting point:

Assume a capacity of `10 l/min`, since this is where the diagram ends.

```
2/22 = 1.0
3/33 = 1.0
4/44 = 1.0
5/55 = 1.0
6/67 = 0.08955223881
7/79 = 0.08860759494
8/91 = 0.08791208791
9/103 = 0.08737864078
10/114 = 0.08771929825
```

We're missing the value for `1 l/min` but to get startet we assume it to similarly be `1/11 = 1.0`.

So what we're ending up with as a starting point could be this:

```
const double cap = 10.0f;  // l/min
const double kf = 11.0f;   // Hz per l/min
FlowSensorProperties sensorConfig = {cap, kf, {1.0f, 1.0f, 1.0f, 1.0f, 1.0f, 0.08955223881f, 0.08860759494f, 0.08791208791f, 0.08737864078f, 0.08771929825f}};
```

thijstriemstra, 2020-05-23:
awesome, thanks for the feedback. i will try it later today and post the results.

sekdiy, 2020-08-12:
Hey @thijstriemstra, did the suggested calibration config work for you?

thijstriemstra, 2020-08-12:
hi @sekdiy it didn't, my own parameters seemed to work better. But I'll revisit this hopefully this summer, otherwise next spring. Feel free to close the ticket though.

sekdiy, 2020-08-12:
That's surprising. Was my calculation off or did the calibration code produce errors?

Closing in case you only get around to answering next year. :) 

Feel free to open again as soon as you're ready!

closed 2020-08-12 by sekdiy