sekdiy/FlowMeter #19: Question on Digiten G1 - Probably FS400A

czuvich, 2020-06-15:
Hi,

First of all, thanks for the great work on this library!  I had a question on sensor calibration in general, and I was hoping you could help since you have a lot of knowledge on this subject.

I'm trying to measure flow/consumption in my garden which uses drip irrigation.  I'm using the following sensor on [Amazon](https://www.amazon.com/gp/product/B00VKAT7EE/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1).  I think the sensor has the same specs as the FS400A; however, the constant value is 4.8 on Amazon which I think is too high anyways (4.5 seems more reasonable).

I've hooked up a few [flow meters](https://www.flows.com/wm-pd-series-low-flow-water-meter/) on the supply line to try and verify the results of the sensor from Amazon.  I also hooked in a [digital flow meter](https://www.amazon.com/gp/product/B06XSNPMBS/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) as well in earlier tests.

Onto my question :)  In 10 minutes, I see around 4.5 gallons consumed at the water flow meters; however, my Digiten sensor is only showing 1.2 gallons or so.  I debugged the output, and I do see around 2 pulses per second which indeed equates to around 4.6L ~ 1.2gal.  

If the Flow Meters are measuring flow accurately, it equates to 1.7 L/m which is within spec of 1-60; however, the error is so high!  Is there something that I'm doing wrong?  Is there a better sensor that yields more accurate results at low flow rates?  Thanks so much!!

sekdiy, 2020-06-16:
Hi @czuvich,

thanks for your feedback (and for using the library). :) 

Drip irrigation in a garden can be tough on sensor linearity, since low flow comes with its own problems. 

> I'm trying to measure flow/consumption in my garden which uses drip irrigation.

What's your water supply? Line pressure, local pump, gravity? Drinking water, well water, rain water? 

The reason why I'm asking is that a flow sensor will always be a restriction to flow, thus measuring the flow rate will inevitably influence it. This is especially true for low pressure and/or low flow systems. 

> In 10 minutes, I see around 4.5 gallons consumed at the water flow meters

Okay, so – as you say – around 1.7 l/min (sorry, metric brain here). 

> my Digiten sensor is only showing 1.2 gallons or so

This *could* very well be sensor nonlinearity, which it will show at both ends of the scale (i.e. for comparatively very low and very high flow rates).

And we don't yet know if any ot the sensors is correct at all... 

The WM-PD has an optional pulse output. Does yours provide one?

The P3 meter isn't a low flow device, so it will have the same linearity problems as the FL-808. 

Do the WM-PD and the P3 agree at low flow?

> If the Flow Meters are measuring flow accurately, it equates to 1.7 L/m which is within spec of 1-60; however, the error is so high!

First of all, the Q&A/review sections on Amazon contain hints about the sensor data (Q = 4.8) being inaccurate and the small bore adding considerable restraint. So I wouldn't expect too much accuracy from the uncalibrated sensor. 

Second of all, the safest method to check would of course be to unhook the drip tubing and take some measurements at different flow rates. 
I recommend the bucket method as described in the wiki (also at higher flow rates in order to verify manufacturer's data). 

And at 1.2/1.7 = 0.706 the error is "only" 30%, which is not *too* bad as long as it's reproducible and stays that way (since a simple correction factor of 1.416 could then take care of it). 
I've seen worse in inexpensive sensors. ;)

> Is there a better sensor that yields more accurate results at low flow rates?

That's a tough question for low flow drip irrigation. 

There are sensors specifically designed for smaller flow rates, e.g. for vending machines. 
But they have even smaller bores and usually aren't very dirt or weather proof. 

If I were in your shoes, I'd try to find the cause for the discrepancy before trying out even more flow sensors (especially since most other sensors also come uncalibrated). 

Also, in my personal experience low flow drip irrigation needs greater care than high flow irrigation  due to dirt, algae, roots, clogging, etc. (so flow rates might vary over time anyway)... 

*Maybe* the WM-PD *with* sensor output could turn out to be a good contender *provided* it gets enough pressure (since it operates on pressure differential).

czuvich, 2020-06-16:
Hi @sekdiy,

Thank you so much for the quick response and great feedback!  Let me try to answer some of these questions.

> What's your water supply? Line pressure, local pump, gravity? Drinking water, well water, rain water?

The supply line is a residential water spigot that is showing 41 PSI.  I have a pressure reader on the line.  The WM-PD is hooked up to the supply line spigot.  I then run 3/4" PVC to my garden.  The raised beds use [1/4" drip irrigation emitters](https://www.dripworks.com/soaker-dripline-6-spacing) which has a 30 PSI regulator sitting in front of them.  I insert the Digiten G1" flow sensor between the pressure regulator and the 1/4" drip (via some adapters and a [poly mainline](https://www.dripworks.com/1-2-polyethylene-mainline-tubing)).  I also have a DC latching valve at each bed to zone irrigation to each bed (I control them via Arduino and LoRaWAN in case you're curious heh).

> Do the WM-PD and the P3 agree at low flow?

No.  The P3 measures a little lower (1 gallon less) in the 10 minute span.  That seems to be in-line with low readings from the Digiten sensors since neither are designed for low flow.

> If I were in your shoes, I'd try to find the cause for the discrepancy before trying out even more flow sensors (especially since most other sensors also come uncalibrated).

That's solid advice.  Based on what I'm reading and your comments, I guess the Digiten sensors (and other variants) are not exactly the most "reliable" sensors.  It's difficult to replicate the exact flow I'm seeing without being hooked up to the drip irrigation system.  I've calibrated via the bucket method with various flow rates by squeezing the line which I know isn't the best method, but the results were within spec of the datasheet.  It's when I saw low flow rates is when the values were _consistently_ skewed.

I have a handful of [3/4" meters from flows.com with pulse output capability](https://www.flows.com/economy-plastic-water-meter-wm-pc-series/) laying around that are calibrated for 1 pulse per gallon.  I haven't tried the pulse output with my uC, but I figure these meters have better reliability and are definitely rated for lower flow (less than 1 L/m it looks like).  I honestly had no idea it would be this difficult to measure water flow!

Side note, this is a hobby project I've been working for over a year, and the water flow is the final (and most cumbersome) piece. Thank you so much for your time and feedback!  It has been extremely useful!

sekdiy, 2020-06-16:
> The supply line is a residential water spigot that is showing 41 PSI.

Okay, that should definitely be enough for any of these sensors to operate. 

> The raised beds use 1/4" drip irrigation emitters which has a 30 PSI regulator sitting in front of them.

Okay. 

> I insert the Digiten G1" flow sensor between the pressure regulator and the 1/4" drip

That shouldn't affect the flow sensor, although I would put the sensor before the pressure regulator in order to maintain 30 PSI (and unrestricted flow) in the drip line. 

> Arduino and LoRaWAN

Cool! :)

> Based on what I'm reading and your comments, I guess the Digiten sensors (and other variants) are not exactly the most "reliable" sensors.

No, in my experience they aren't. But with regular maintenance (opening them up and cleaning the impeller, shaft and bores) they are good enough for irrigation purposes. 

To be fair, non-commercial garden irrigation with municipal water is probably one of the the least demanding applications for a flow sensor. 

> the results were within spec of the datasheet. It's when I saw low flow rates is when the values were consistently skewed.

Well, as long as they are skewed in the same direction, the `FlowMeter` library allows to compensate for it via it's calibration option. 

If you limit your system capacity to a relatively low (maximum) flow rate, your can provide correction factors for *low* or even *very low* flow rates. 

> the water flow is the final (and most cumbersome) piece

Wait until you try to reliably sense soil moisture and PH in multiple locations – wirelessly. :D

czuvich, 2020-06-16:
> That shouldn't affect the flow sensor, although I would put the sensor before the pressure regulator in order to maintain 30 PSI (and unrestricted flow) in the drip line.

That is a really good point!  Thank you!

> To be fair, non-commercial garden irrigation with municipal water is probably one of the the least demanding applications for a flow sensor.

True.  I plan on demonstrating my product for small growers/nurseries, but for now it's in my 1 acre plot :)  I could see why it's not as demanding in this industry simply due to infrastructure cost (not just the sensor.. but everything else).

> If you limit your system capacity to a relatively low (maximum) flow rate, your can provide correction factors for low or even very low flow rates.

The only caveat to that I think is leak detection.  I've had instances where the irrigation lines pop off since they aren't glued together.  It'd be nice to maintain the proper accuracy at the higher flow rates, too.

> Wait until you try to reliably sense soil moisture and PH in multiple locations – wirelessly. :D

Already done ;)  I sense soil moisture and soil temperature.  I attached a picture of one of the pages on my mobile app that shows the water consumption per zone  as well as how it affected the soil moisture and temperature.  I use this as data to help automate opening/closing the valve for more efficient watering and overall healthier plants (roots are stronger if the soil is dried out between watering).

Anyways, thank you so much for the tips and comments. I really do appreciate it.

[image]

sekdiy, 2020-06-17:
> The only caveat to that I think is leak detection.

If a popped line always means maximum flow (i.e. predetermined upper limit), wouldn't that be easy to detect?

> I sense soil moisture and soil temperature.

Can you reveal your sensor type(s)?

czuvich, 2020-06-17:
> If a popped line always means maximum flow (i.e. predetermined upper limit), wouldn't that be easy to detect?

Good point.

> Can you reveal your sensor type(s)?

I use the [Vegetronix VH400](https://www.vegetronix.com/Products/VH400/); however, temperature has an effect on the readings which is why I also measure temperature.  I've also noticed installation is critical.  I've tried half a dozen different sensors, but there are still a few more I'd like to try that are not available in the US yet.

closed 2020-07-19 by czuvich