sekdiy/FlowMeter #21: .getCurrentVolume  reset every request

deckchairboy, 2020-08-05:
I am having some trouble monitoring pumped volumes using .getCurrentVolume .

This is a section of my code that runs in a subroutine called from the loop as needed.

 unsigned int targetTotal =  Ml; //millilitres to be dispensed, Ml is passed when the dispense function is called.
  flowMeter.reset();
  double aliquot = 0;
  
  while ((flowMeter.getCurrentVolume() * 1000) < targetTotal) {       //*1000 to get Millilitres,,
    flowMeter.tick(flowMeterPeriod);             //flowMeterPeriod is 1000
    aliquot = flowMeter.getCurrentVolume() * 1000;
    lcd.setCursor(12, 3);
    lcd.print(aliquot);
    Serial.println(aliquot);
    if (digitalRead(setPin) == LOW) {  //hardware switch to cancel dispensing
      break;
    }

    currentTime = millis();  
    if ((currentTime - startTime) > timeOut) {  //Time-out function timeOut set by volume of requested dispense
      lcd.setCursor(11, 3);
      lcd.print("T");
      timedOut = true;
      break;
    }
    delay(flowMeterPeriod);  // blocking behaviour ok in this function?
  }

I expect to see the value of current volume in the variable "aliquot"  incremented every second giving a total of the volume dispensed during this dispense cycle, after the reset instruction, ie counting from 0 up.

What I actually see is the volume dispensed in a second displayed every second and not incrementing. ie 5.1, 5.1, 5.3, 5.0 etc.

The totalVolume function works as anticipated and shows the total volume incremented by the dispensed volume at the end of the dispense function.

I am using an Arduino nano.

I've been stuck on this for a while but apologise for any stupidity on my own part in advance

sekdiy, 2020-08-12:
Hi @deckchairboy,

I'm afraid the documentation on `.getCurrentVolume()` might be misleading. 

In the source header it says:

```c++
double getCurrentFlowrate();    //!< Returns the current flow rate since last reset (in l/min).
double getCurrentVolume();      //!< Returns the current volume since last reset (in l).
```
The bit `since last reset` might be… _wrong_ and should instead read `since last tick`. 

Additionally, there have been requests for resetting the total volume and flow rate during `reset()` from the beginning (see #1). 

Such a change would enable you to simply use `getTotalVolume()` instead.

I defended my initial design decision to preserve total stats in order to make a `FlowMeter` instance consistent (i.e. never reduce its totals inbetween readouts).

But maybe nobody needs this feature... while everyone would expect everything to be reset during a `reset()`. 

Anyways, the feature you'd need within your subroutine isn't implemented yet, so you'd have to accumulate the volume yourself (e.g. via `aliquot += flowMeter.getCurrentVolume() * 1000;`). 

I'll correct the documentation in order to reflect the actual behaviour.

Please let me know if the suggested accumulation within your subroutine works for you and what you think about preserving total volume and flow rate during a `reset()`. 

As an alternative I could introduce a _third_ statistical value (e.g. `getVolumeSinceLastReset()`) in order to support accumulation inbetween resets without loosing consistent totals.

deckchairboy, 2020-08-12:
Dear Sebastian

Your idea of using an accumulation within the subroutine works just fine of
course but I have a feeling this may become a thorn in your side if you
give in to pressure to modify your software.

Your library is elegant, minimal and useful (Thank you!)  it just has this
small anomaly.  Life is short - I would just change the documentation to
highlight the real purpose of getCurrentVolume with regard to the Tick
event and also point out that a valueSinceLastReset can be created with an
incremented global variable manually reset immediately after a flowmeter
reset call.

I support your standpoint that a call to reset the flowmeter should not
reset the total in the session.   I need the total volume during the
session as well as the volume per each dispense event.

It's a great piece of work, it doesn't need changing, just a tweak to the
documentation and get on with life, love, family, friends....... and a new
project!

regards
Paul

sekdiy, 2020-08-13:
> Your library is elegant, minimal and useful (Thank you!)

Woah! :)

> I would just change the documentation to highlight the real purpose of getCurrentVolume

Done (see above).

> I support your standpoint that a call to reset the flowmeter should not reset the total in the session.

Yes, that's something I will not change.

I'd rather explain my reasoning to everyone who asks... ;)

> It's a great piece of work

Stop! :D

> it doesn't need changing

This will be discussed further in issue #1

I might add another set of stats instead of changing the ones available now.

While this might _seem_ like feature bloat, it's actually something that's been requested multiple times now. 

So it could be considered relevant...

Thanks or your input, @deckchairboy!

closed 2020-08-13 by sekdiy