sekdiy/FlowMeter #23: problem with library

Sayed88, 2021-10-10:
**Description**
there are a problem with the reset of meter i have a Yf s402 sensor i have make some calibration for it but it in first filling gives required quantity second it is wrong with minus

**Reproduction**
Steps to reproduce the behavior:
1. Connect '...'
2. Measure '...'
3. Error '...'

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware (please complete the following information):**
 - Board type
 - Sensor type
 - Pins used

**Software (please complete the following information):**
 - OS
 - IDE
 - Other libraries

**Minimal example to reproduce (please complete the following information):**
 - Code to reproduce
 - Wiring to reproduce

**Additional context**
Add any other context about the problem here.

sekdiy, 2021-10-12:
Hello,

thanks for reporting the issue.

I'm afraid I need more information in order to understand the nature of the problem.

I understand that you would like to use the `reset` function.
Could you please describe what you would like to achieve and what you have tried so far?

Thanks!

sekdiy, 2021-12-15:
If you have new information, please reopen this issue.

closed 2021-12-15 by sekdiy