sekdiy/FlowMeter #24: Gicar flowmeter unstable doses

tasosman, 2022-01-07:
**Description**
A clear and concise description of what the bug is.

**Reproduction**
Steps to reproduce the behavior:
1. Connect '...'
2. Measure '...'
3. Error '...'

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware (please complete the following information):**
 - Board type
 - Sensor type
 - Pins used

**Software (please complete the following information):**
 - OS
 - IDE
 - Other libraries

**Minimal example to reproduce (please complete the following information):**
 - Code to reproduce
 - Wiring to reproduce

**Additional context**
Add any other context about the problem here.
Hi I tried so hard to calibrate my flowmeter from gicar.has 1 mm Ziegler and I can’t set it up properly all the time so different values .if it is possible can we see the pulses from the flowmeter?that it can help much more 
[image]

sekdiy, 2022-01-07:
Hi,

> can we see the pulses from the flowmeter

You can't see the pulses directly, because the serial output would be too slow.

In your situation it might be helpful to use the function `getCurrentFrequency()`.

It provides the number of pulses per second – so not every single pulse, but the sum over 1s.

Could you please provide your code (or a minimal example) so that I can understand your issue a little better?

The Gicar flow meters should work.

tasosman, 2022-01-07:
I use your example,I have one button connect with the one valve so when I start the pump and valve is open the flowmeter start counting if I close the valve I give the command for meter reset but is look like missing pulses all the time.I have change the delay time from 1000 you have to your example to 25 so it can count more.

tasosman, 2022-01-07:
I have connect also one 10k resistor between + and signal

sekdiy, 2022-01-10:
Could you please state the model/type of your flow meter?
Gicar made a lot of different models over the years.

In case you're using the `Calibration.ino` example (which I recommend for your situation), the time constant does not cause a 'delay', but a period of 'active waiting'.
No data is being missed during such a delay or waiting period, since the microcontroller does the counting with the help of 
dedicated hardware.

If you modified the library's example code, please consider posting it here.
If you don't want to post it here, please provide a minimal example that still reproduces the issue.

By the way: which microcontroller are you using and which pin do you connect the flow meter to?

> when I start the pump and valve is open

Have you tried simply spinning the flow meter (e.g. by blowing into it) without anything else attached or running?

> I have change the delay time from 1000 you have to your example to 25 so it can count more.

I'm not sure I understand.

If you reduce the time constant in the example sketches from 1000 to 25, the reporting interval is reduced by a factor of 40.
This would mean that the likelihood of missed pulses increases 40-fold, because the calculations/statistics and the serial output will now be done 40 times more often.

What would you like to achieve by going with 25, i.e. have you tried longer periods instead of shorter ones?

tasosman, 2022-01-16:
hi i have the 1mm with 2 magnets so dose per pulse is 0,47cm3
pulses per liter 2128,
flow rate 1400 cc/min
 controller i use esp8266 nodemcu lua pin D8
my code for flow meter is(is not all my code i sent you the part of flowmeter if you want i can sent it you all)

//void ICACHE_RAM_ATTR MeterISR();
// enter your own sensor properties here, including calibration points
FlowSensorProperties MySensor = {60.0f, 33.33f, {1, 1, 1, 1, 1, 1, 1, 1, 1, 1}};
// connect a flow meter to an interrupt pin (see notes on your Arduino model for pin numbers)
FlowMeter *Meter;

//flowmeter
// define an 'interrupt service handler' (ISR) for every interrupt pin you use
void  ICACHE_RAM_ATTR MeterISR(){
    // let our flow meter count the pulses
    Meter->count();
}

// set the measurement update period to 1s (1000 ms)  //i using 25 because i want to know very fast the results because the ml is max 65 and i want to know very fast the results
const unsigned long period = 25;

void setup() {
  Serial.begin(9600); 
  Serial.println("name");
  Serial.println("esp1.2v ");
  tft.begin();
  tft.fillScreen(BLACK);
  tft.setRotation(1);
   pinMode(A0, INPUT);
   int sensorVal = digitalRead(D1);

//flowmeter  
  // get a new FlowMeter instance for an uncalibrated flow sensor on pin 3
    Meter = new FlowMeter(digitalPinToInterrupt(D8), MySensor, MeterISR, RISING);

}
void loop() { 

int yF =((float)Meter->getCurrentFlowrate()*100);   //for flow graph ml/s

 tft.drawLine(20+x*5,119,20+x*5,120-(yF/50),ORANGE);   //flow/min

//flowmeter 

// wait between output updates
    delay(period);

    // process the (possibly) counted ticks
    Meter->tick(period);

    // output some measurement result
    Serial.print("Currently ");
  Serial.print(Meter->getCurrentFlowrate());
  Serial.print(" ml/min, ");
  Serial.print(Meter->getTotalVolume()*100);
  Serial.println(" ml total.");
  
  tft.setTextSize(2);
     tft.setTextColor(WHITE,BLACK);
     tft.setCursor(210,60);
      tft.setTextSize(1);
   tft.print("ml/min");
   
    tft.setTextSize(2);
   tft.print(Meter->getCurrentFlowrate()*100);
  
   tft.setCursor(250, 35);
   tft.setTextSize(2);
   tft.print(Meter->getTotalVolume()*100);
   //tft.println("ml");
    //

sekdiy, 2022-01-29:
Sorry for the late reply.

I edited your comment to improve readability (code formatting).

There are a couple of things that raise questions for me.

First of all, this isn't really a 'minimal' example, right? ;)

Have you tried leaving the button, the valve and the TFT out of the circuit in order to sort out the flow sensor issues?
By that I mean physically disconnecting everything except the flow sensor and the serial monitor connection. 
At this point, the issue might still be mechanical, electrical or in the math (so, basically, everywhere)…

Second of all, you still haven't given the model/type of your flow sensor, but you use a k-factor of `33.33f` in your setup.

With 2128 pulses/l and assuming a flow rate of 1 l/min, you'll have ≈ 35,47 pulses/s.
Let's assume the pulse rate to be linear to keep things simple. 
Thus, the k-factor calculates as `kf = 35.47f` (1/s) / (l/min).

How did you arrive at 33.33 without measuring?

Third of all, what is the output data/graph you get?

Thus far you have just stated that the values don't look right, but you haven't provided any actual output data.
What's your actual flow rate, i.e. are you sure you're within the spec of 32 ml/min to 1.4 l/min?

One additional thing you could look at: 
Depending on the compiler you use, `yF =((float)Meter->getCurrentFlowrate()*100)` *could* mean converting `double` to `int`, then to `float`.

closed 2022-07-18 by sekdiy