sekdiy/FlowMeter #27: YF-DN32

guestisp, 2023-08-17:
Anyone using the YF-DN32 flow sensor ? 
I can't find any reliable specs, some sites tell k=0.45*Q, other 4.5Q, other 0.45Q = l/m and so on

which one is the correct value ?

atm, i'm using the following:
```
  FlowSensorProperties YF_DN32 = {120.0f, 0.45f, {1, 1, 1, 1, 1, 1, 1, 1, 1, 1}};
```

but don't know if correct

guestisp, 2023-08-19:
obviously i've tried with 0.45 but returned flow is way too high to be real. something plausible is near 0.70

sekdiy, 2023-08-19:
It's possible you are the first user to report back with such a YF-DN32 sensor.

I've never come across one myself, so can't verify.

There are a couple of discussions in (now closed) issues about proper calibration for various sensor types.

The meaning of the `kFactor` in this library is defined per the equation: `pulses/s = kFactor * liters/min` (which converts to `kFactor =  pulse/s per liters/min`).

The expression `0.45Q = l/m` usually means `kFactor = 0.45` *at a flow rate* that the manufacturer recommends/specifies the sensor for.

In reality the true `kFactor´ hugely depends on flow rate (and thus pulse rate), which is an expression of sensor nonlinearity.

That's what the array of `mFactor` values is for (all still *unity* in your instance).

Essentially, the `kFactor` of the YF-DN32 *might* very well be 0.45 at *some* flow rate, but it turns out to be more near 0.70 at *your* flow rates…

Long story short: did you run the calibration procedure as per the [wiki page](https://github.com/sekdiy/FlowMeter/wiki/Calibration)?

guestisp, 2023-08-19:
no, i didn't run the calibration because i've thought it was for accurate measurament. in my case, knowitng 1900 lites/h or 1800 liters/h is the same , but i'm getting 3200 liters/h (obviously, this is not the same)

guestisp, 2023-08-19:
Also, i don't have anything able to reach even near the sensor capacity (120 liters/min)

guestisp, 2023-08-19:
I know 2 things:

the bigger model, DN40, has the same K factor (0.45). This is strange, larger wheel, same k factor.....
the smaller model, DN25, has k factor 1, so i would expect something in the middle for the middle sensor (DN32): (1.0 + 0.45)/2 = 0.725

Results with 0.725 looks correct, but i don't know if they are correct.

sekdiy, 2023-08-21:
> i didn't run the calibration because i've thought it was for accurate measurament

It also allows you to discover the sensor properties in case they are unknown or implausible. 

Maybe I should add this topic to the documentation.

> i don't have anything able to reach even near the sensor capacity (120 liters/min)

Can you give an estimate of your actual flow rate?

All the things you need for such an estimate are a bucket, a watch and a minute of your time – and that's also the first step of the calibration process. ;)

> Results with 0.725 looks correct, but i don't know if they are correct.

The calibration process will sort this out – and it's quicker than further research or more guesswork.…

guestisp, 2023-08-22:
not really

5 liters in 10secs are 5/10 = 0.5 liters per second that are 0.5*60=30 liters/min that are 30*60=1800 liters per hour

sekdiy, 2023-08-22:
> 5 liters in 10secs are 5/10 = 0.5 liters per second that are 0.5 * 60 = 30 liters/min

Yep, sorry, 30 l/min and 0.5 l/s (not 36 and 0.6), my bad. 
Still far from 1800 l/s. ;)

Okay, so you're in the lower quarter (30/120 = 0.25) of your sensor's range.
That might explain *some* of the deviation.

You didn't mention your application, but what flow range do you intend to use, e.g. what's your lowest and what's your highest expected flow?

You could measure Q factors at e.g. 1 l/min (5min to fill a 5l bucket), 3 l/min (100s to fill a 5l bucket), 6 l/min (50s to fill a 5l bucket), and so on – up to 30 l/min.
This might tell you the true behaviour of your sensor at these relatively low flows.
Note that taking these measurements basically correspond to … a calibration run. :)

guestisp, 2023-08-22:
1800 liters HOUR

i'm planning to use the sensor to measure the real flow of some unknown water pumps i have

sekdiy, 2023-11-27:
I'm closing this issue due to inactivity.

If you have new information, please just reopen it.

closed 2023-11-27 by sekdiy