sekdiy/FlowMeter #28: Compile fail with Nano 33 IOT and MKR 1010

mackrasp, 2024-01-09:
**Description**
FlowMeter example Simple.ino does not compile for Nano 33 IOT (using pin 2 or 3) and MKR 1010 (using pin 4) boards.
Compiles/Verify correctly for Uno and Nano boards.

**Reproduction**
Steps to reproduce the behavior:
1. Connect '...'
2. Measure '...'
3. Error '...'
4. FQBN: arduino:samd:nano_33_iot
Using board 'nano_33_iot' from platform in folder: /Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13
Using core 'arduino' from platform in folder: /Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13

Detecting libraries used...
/Users/mack/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_NANO_33_IOT -DARDUINO_ARCH_SAMD -DCRYSTALLESS -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8057 -DUSBCON -DUSB_MANUFACTURER="Arduino LLC" -DUSB_PRODUCT="Arduino NANO 33 IoT" -I/Users/mack/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/Users/mack/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/ -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/api/deprecated -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/api/deprecated-avr-comp -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/variants/nano_33_iot /private/var/folders/5v/dc3kh_4136g2j560dgjg3vww0000gn/T/arduino/sketches/C68A8D46ABA33F055F35B22017641534/sketch/Simple.ino.cpp -o /dev/null
Alternatives for FlowMeter.h: [FlowMeter@1.2.0]
ResolveLibrary(FlowMeter.h)
  -> candidates: [FlowMeter@1.2.0]
/Users/mack/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_NANO_33_IOT -DARDUINO_ARCH_SAMD -DCRYSTALLESS -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8057 -DUSBCON -DUSB_MANUFACTURER="Arduino LLC" -DUSB_PRODUCT="Arduino NANO 33 IoT" -I/Users/mack/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/Users/mack/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/ -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/api/deprecated -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/api/deprecated-avr-comp -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/variants/nano_33_iot -I/Users/mack/Documents/Arduino/libraries/FlowMeter-master/src /private/var/folders/5v/dc3kh_4136g2j560dgjg3vww0000gn/T/arduino/sketches/C68A8D46ABA33F055F35B22017641534/sketch/Simple.ino.cpp -o /dev/null
/Users/mack/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_NANO_33_IOT -DARDUINO_ARCH_SAMD -DCRYSTALLESS -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8057 -DUSBCON -DUSB_MANUFACTURER="Arduino LLC" -DUSB_PRODUCT="Arduino NANO 33 IoT" -I/Users/mack/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/Users/mack/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/ -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/api/deprecated -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/api/deprecated-avr-comp -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/variants/nano_33_iot -I/Users/mack/Documents/Arduino/libraries/FlowMeter-master/src /Users/mack/Documents/Arduino/libraries/FlowMeter-master/src/FlowMeter.cpp -o /dev/null
Generating function prototypes...
/Users/mack/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_NANO_33_IOT -DARDUINO_ARCH_SAMD -DCRYSTALLESS -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8057 -DUSBCON -DUSB_MANUFACTURER="Arduino LLC" -DUSB_PRODUCT="Arduino NANO 33 IoT" -I/Users/mack/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/Users/mack/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/ -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/api/deprecated -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/api/deprecated-avr-comp -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/variants/nano_33_iot -I/Users/mack/Documents/Arduino/libraries/FlowMeter-master/src /private/var/folders/5v/dc3kh_4136g2j560dgjg3vww0000gn/T/arduino/sketches/C68A8D46ABA33F055F35B22017641534/sketch/Simple.ino.cpp -o /private/var/folders/5v/dc3kh_4136g2j560dgjg3vww0000gn/T/4284101656/sketch_merged.cpp
/Users/mack/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /private/var/folders/5v/dc3kh_4136g2j560dgjg3vww0000gn/T/4284101656/sketch_merged.cpp
Compiling sketch...
/Users/mack/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_NANO_33_IOT -DARDUINO_ARCH_SAMD -DCRYSTALLESS -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8057 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino NANO 33 IoT\"" -I/Users/mack/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/Users/mack/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/ -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/api/deprecated -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/api/deprecated-avr-comp -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/variants/nano_33_iot -I/Users/mack/Documents/Arduino/libraries/FlowMeter-master/src /private/var/folders/5v/dc3kh_4136g2j560dgjg3vww0000gn/T/arduino/sketches/C68A8D46ABA33F055F35B22017641534/sketch/Simple.ino.cpp -o /private/var/folders/5v/dc3kh_4136g2j560dgjg3vww0000gn/T/arduino/sketches/C68A8D46ABA33F055F35B22017641534/sketch/Simple.ino.cpp.o
Compiling libraries...
Compiling library "FlowMeter"
/Users/mack/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_NANO_33_IOT -DARDUINO_ARCH_SAMD -DCRYSTALLESS -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8057 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino NANO 33 IoT\"" -I/Users/mack/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/Users/mack/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/ -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/api/deprecated -I/Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/api/deprecated-avr-comp -I/Users/mack/Library/Ard [... cut ...]
Using library FlowMeter at version 1.2.0 in folder: /Users/mack/Documents/Arduino/libraries/FlowMeter-master 
exit status 1

Compilation error: exit status 1

**Hardware (please complete the following information):**
 - Board type Nano 33 iot
 - Sensor type
 - Pins used

**Software (please complete the following information):**
 - OS 12.7.2
 - IDE2.2.1
 - Other libraries

**Minimal example to reproduce (please complete the following information):**
 - Code to reproduce Simple.ino
 - Wiring to reproduce

**Additional context**
Add any other context about the problem here.

sekdiy, 2024-01-09:
Hi,

Your error trace reveals the following potential error source:

```sh
FlowMeter.cpp:18:83: error: no matching function for call to 'attachInterrupt(unsigned int&, void (&)(), uint8_t&)'
attachInterrupt(this->_pin, this->_interruptCallback, this->_interruptMode);
```

Please try patching the file "FlowMeter.cpp" in line 18 as follows:

```diff
-        attachInterrupt(this->_pin, this->_interruptCallback, this->_interruptMode);
+        attachInterrupt(digitalPinToInterrupt(this->_pin), this->_interruptCallback, (PinStatus) this->_interruptMode);
```

The key changes here are the introduction of the `digitalPinToInterrupt()` function and the explicit cast to `(PinStatus)`.
You can try one, the other or both.

A similar issue had been identified in [#22](https://github.com/sekdiy/FlowMeter/issues/22#issuecomment-836679534), but so far there doesn't seem to be a universal solution that works for every platform.

The current documentation is available [here](https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/#_syntax.
The "Syntax" section *recommends* the use of `digitalPinToInterrupt()`, but still keeps it ambiguous.
There's also an "Example Code" section that you could try in order to identify if you can reproduce the issue without this library.

mackrasp, 2024-01-10:
Hi,
Thanks.
The use of the explicit cast to (PinStatus) allows the code for Nano 33 IOT and MKR WIFI 1010 boards to be compiled and the Simple.ino program executes OK on these platforms.

However adding this code will not compile for the Uno and Nano boards.
So there is now a solution all be it not a universal solution.
The use of code digitalPinToInterrupt(this->_pin),   does not appear to affect the compilation.

Reply to: sekdiy/FlowMeter ***@***.***>
Date: Wednesday, 10 January 2024 at 4:08 am
To: sekdiy/FlowMeter ***@***.***>
Cc: Mack Richards ***@***.***>, Author ***@***.***>

sekdiy, 2024-01-10:
> The use of the explicit cast to (PinStatus) allows the code for Nano 33 IOT and MKR WIFI 1010 boards to be compiled and the Simple.ino program executes OK on these platforms. 

Glad to hear it. :)

> However adding this code will not compile for the Uno and Nano boards. So there is now a solution all be it not a universal solution. 

Like I mentioned, this issue came up before and it's sad that –still – no truly portable solution seems to be available.

> The use of code digitalPinToInterrupt(this->_pin), does not appear to affect the compilation. 

Okay.

closed 2024-01-10 by sekdiy